RockRaidersInc/Electrical #47: Design/Source fan guards

cmcgowan27, 2019-09-08:
We've had issues with wires and fingers getting into the fans

Strat70, 2019-09-27:
Found a guard that seems like it would work, design available on amazon and on a website for bulk orders. Figure ordering some would be cheaper and less work than  3D printing them
-Amazon (higher part cost): https://www.amazon.com/Accessories-Plastic-Guard-Size-piece/dp/B00M28WHQS/ref=sr_1_1?keywords=92mm+Plastic+Fan+Guard&qid=1569613320&s=gateway&sr=8-1
-Other (higher shipping cost): https://www.productsforautomation.com/gardtec-92mm-plastic-fan-guard-sc92p10-p/SC92-P10.htm?gclid=Cj0KCQjw5rbsBRCFARIsAGEYRwe74oWWL-tGW0mVA80-0hhSPhdU1oRt1y-CCKCPgFAmnmMbuerYBvoaAsUUEALw_wcB

Depending on how many guards we need, one may be better over the other . I am also assuming we don't need guards with filters, but there are some options for that as well

cmcgowan27, 2019-09-28:
We'd probably be getting 6, no filter needed. Do you think you could try to find one from an approved vendor? There should be a pinned post in the electrical channel. If not, don't worry about it. There are ways of making small purchases from unapproved vendors without too much hassle.

Strat70, 2019-10-04:
Here's the link from an approved vendor:
https://www.digikey.com/products/en/fans-thermal-management/fans-finger-guards-filters-sleeves/221?k=fan+guard&k=GardTec+SC92-P10+92mm+Plastic+Fan+Guard&pkeyword=fan+guard&sv=0&v=1997&sf=0&FV=ffe000dd&quantity=&ColumnSort=0&page=1&pageSize=25

cmcgowan27, 2019-10-04:
Looks good. Closing this issue.

closed 2019-10-04 by cmcgowan27