RockRaidersInc/Electrical #56: Design power cutoff for wireless E-Stop

cmcgowan27, 2019-09-09:
Right now, we use a latching relay in series with our regular E-Stop. When you activate it, it latches open and opens the onboard E-Stop circuit controlling the relay. The relay then needs to be manually reset by pressing a button on the rover.

If you think this is already the best solution, also take a look at whether our hardware could be improved, or whether we can make something that doesn't require a manual reset.

MBacon15, 2019-10-26:
The hardware should be sufficient. I think mounting the relay itself to the future distribution board is a good idea. 

The main QoL issue I've been looking at is the manual reset. I'd like to look into a dual wireless/manual reset solution using an Ultra Low Power Arduino like this, [Mini Ultra 8 MHz](https://www.rocketscream.com/blog/product/mini-ultra-8-mhz-arduino-compatible/). 
Maybe using one powered with a couple of batteries to let the board switch the relay on and off again, as opposed to requiring power from the main battery.

cmcgowan27, 2020-02-09:
Power cutoff has been integrated into the distribution PCB using same relay

closed 2020-02-09 by cmcgowan27