RockRaidersInc/Electrical #61: Select lane detection cameras

cmcgowan27, 2019-09-10:
We need USB cameras to mount to the side of the rover to see the lanes. 1080p should be fine.

cmcgowan27, 2019-09-25:
This is our camera: https://www.amazon.com/Creative-Live-Sync-720P-Webcam/dp/B07P8X6PHS?th=1

cmcgowan27, 2019-09-26:
There were some concerns that the chosen camera might not have a fixed focus. Reopening. Maybe we want to look for ones with a really wide field of view.

daweim0, 2019-09-26:
I'm thinking we should get this camera - https://www.amazon.com/Logitech-Desktop-Widescreen-Calling-Recording/dp/B004FHO5Y6/ref=sr_1_1?keywords=logitech+270&qid=1569526024&s=pc&sr=1-1 - it has manual focus and generally well respected. The $11 one posted earlier seems to have bad dynamic range (looking at a pc world review)

cmcgowan27, 2019-09-26:
I believe we had passed over that one because it only has 60 degrees of FOV and because it explicitly said that it doesn't support Linux.

cmcgowan27, 2019-09-28:
Gonna get this one and see if we like it: https://www.amazon.com/Creative-Live-Sync-720P-Webcam/dp/B07P8X6PHS?th=1

daweim0, 2019-09-28:
UPDATE: we don't know if autofocus will be an issue. We don't want the camera auto-focusing while driving down the course, but it might not in practice. We currently have a  pretty good USB webcam (with auto-focus) in the closet. We should use that until we figure out if auto focus is a problem or not. This issue will be re-visited later.