RockRaidersInc/Electrical #65: Create an external reset button for the Jetson Nano

cmcgowan27, 2019-09-15:
The button should go in series with the power supply, and cut off the power when pressed.The circuit needs to be able to switch 2A.

Since buttons usually have the opposite behavior, you might need to put a transistor in series with the power supply and use a pull up resistor to keep it on. Then just have the button hold the gate of the transistor at ground when pressed.

cmcgowan27, 2019-09-17:
I just got a nice rocker switch that we could use for this

MBacon15, 2019-09-25:
Note for the switch: Sides are isolated from each other, when looking at the switch from the sides the center contact is connected to the contact on the protruding part of the switch.

MBacon15, 2019-09-28:
Switch spliced into power line for Jetson. Switch will eventually be labeled. Top depressed = On, Bottom depressed = Off

closed 2019-09-28 by MBacon15